PRBonn/bonnet #5: nodes.yaml is missing?

susu3621, 2018-03-17:
This project is very attractive! But when I tried it, the nodes.yaml seems missing.
Is this file missed?
=====log=====
Successfully created log directory: log
Unable to create network.
Invalid yaml file city_512/nodes.yaml
=====

tano297, 2018-03-17:
Hello,

The reason why it is missing is because the folder provided is a log folder, and in the case you are using you need the frozen model. This is done this way by design, as the "nodes.yaml" is only necessary when using a frozen model, and its contents can be automatically generated to reduce user input. 

To learn how to freeze the log folder, please refer to the python section of the readme! 

Let me know if you need further help, otherwise I will close this issue.

Thanks!

susu3621, 2018-03-17:
Thanks for your quick reply.
Have confirmed the code works perfect after I freeze the model.
Thanks a lot. I close this issue.

closed 2018-03-17 by susu3621

tano297, 2018-05-07:
I can freeze some for you if you tell me which, but the idea of keeping the logs separate from the frozen models is that as we add backends, or if we do specific optimizations to the graph during freezing and then the backends change, the frozen models break. 

I strongly recommend you give the docker container a look, it contains all the dependencies, so it should be a matter of minutes to get the freezing going.

mfe7, 2018-05-08:
@tano297 Thanks for that suggestion. I'll try out the docker

chenyumingx, 2018-05-21:
I have encountered the same problem,"Invalid yaml file persons_512/nodes.yaml"
but I do not fully understand your answer, and **do not know how to solve it**. Can you answer in more details? Thank you very much!

AI-liu, 2018-06-05:
export PYTHONPATH=/usr/local/lib/python3.6/site-packages/cv2:$PYTHONPATH
python3 cnn_freeze.py -p /tmp/path/to/pretrained -l  /tmp/path/to/log

maybe you missed  -l  /tmp/path/to/log