PRBonn/bonnet #6: Import Error cv2 using docker image

srcolinas, 2018-03-31:
Hi, your repo seems very useful and interesting, I thank you for that.

However, I got the following error trying to make it segment an image:
```
Traceback (most recent call last):
  File "./cnn_use.py", line 30, in <module>
    import cv2
ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type
```
I use the docker image as you explain in the main README.md and try the following comand:
`~/bonnet_wrkdir/train_py$ ./cnn_use.py  -p city_512/acc/ -i ../test.jpg`
I had the following folder structure and got the -city_512 from your download link:
```
- deploy_cpp/
- train_py/
    - city_512/
    - test.jpg
```
What can I do to make it work? 
Thanks for your help!

bgheneti, 2018-04-10:
I think this needs to run in an environment with python3, not python2, which ros is setup with by default.

https://stackoverflow.com/questions/43019951/after-install-ros-kinetic-cannot-import-opencv

tano297, 2018-04-12:
Hi, sorry for the long delay, I've been swamped (and sick). As @bgheneti pointed out, it is a problem with the ros install.

I started modifying the docker file to fix this, but I realized that it made no sense because I will have a major overhaul coming soon to support the latest tensorflow (1.7), the latest tensorrt (4), and the latest versions of cuda and cudnn (9, and 7).

Watch the repo for updates.

tano297, 2018-04-16:
I've made significant changes to the repo and the docker image to work with tensorflow 1.7 and tensorrt 3.0.4, which work well with cuda9 and cudnn7, and in the meantime I also fixed the python cv2 library conflict with ROS. It should be solved now, feel free to reopen if you need any more help.

closed 2018-04-16 by tano297