PRBonn/bonnet #7: make failed

3bobo, 2018-04-17:
I installed tensorflow under an env with anconda , so I try to make bonnet under the same env , but there is an error :

ImportError: "from catkin_pkg.package import parse_package" failed: No module named catkin_pkg.package
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/zqk/anaconda2/envs/tensorflow/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/zqk/detection_code/bonnet/deploy_cpp/build/catkin_tools_prebuild/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:151 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)
 
when I run `python -V`,it is:
`Python 2.7.14 :: Anaconda, Inc.`
but after I exit the env and make the bonnet , it failed again.
can you plz help me? thanks

tano297, 2018-04-18:
Hi there,
The C++ inference library has no relation with the python environment. You are having some catkin tools problems. Please attach a list of the commands you are executing and a description of what you are trying to do so that I can help you further.
Thanks

3bobo, 2018-04-18:
`sudo pip install catkin_tools trollius`
`sudo apt install libboost-all-dev libyaml-cpp-dev`
`cd bonnet/deploy_cpp`
`catkin init`
`catkin build bonnet_standalone`
I have installed ros-kinetic and tensorflow with anaconda, and I want to deploy the C++ code  with ros   by what the [readme](https://github.com/Photogrammetry-Robotics-Bonn/bonnet/blob/master/deploy_cpp/README.md) had told.thanks for your help

tano297, 2018-04-18:
Hi again,

- The tensorflow install for the C++ inference is different from the python install, so it doesn't matter what you do in your python env. You need to follow this part of the [README](https://github.com/Photogrammetry-Robotics-Bonn/bonnet/tree/master/deploy_cpp), for the deployment part of the pipeline. You need to use tensorflow_cc to make a shared library (.so) so that you can link against it in the ROS app.

- The build for ros should be done inside an existing catkin workspace, where you have your other ros nodes, so the `catkin init ` part is not necessary.

- To build the ros part, you should run 
```sh
$ catkin build bonnet_run # not bonnet_standalone
```

- There is no such thing as installing ros in a virtual environment, as far as I am concerned, please follow the steps [here](http://wiki.ros.org/kinetic/Installation).

- If you have an nvidia gpu, you can use the tensorRT backend instead of tensorflow_cc, and you will see improvements in speed. It is not only faster, but also easier to install.

In summary, you need to install ROS, tensorflow_cc (or tensorRT), put the code inside an existing catkin workspace (or create a new one if you don't have any other nodes), and then build bonnet_run

tano297, 2018-04-19:
I'm going to close this because I have not heard of this being a problem in any other case, and the docker container is working properly, so it can be used as well. For any other doubts you can send me an email to avoid spamming everybody.
Thanks!

closed 2018-04-19 by tano297