PRBonn/bonnet #8: [build] Error: Unable to find source space `/bonnet/src`

AhmedElsafy, 2018-04-19:
trying to build I get the following 

Source Space:      [missing] /bonnet/src
Log Space:         [missing] /bonnet/logs
Build Space:       [missing] /bonnet/build
Devel Space:       [missing] /bonnet/devel
Install Space:      [unused] /bonnet/install
DESTDIR:            [unused] None

tano297, 2018-04-19:
If you are getting that error it's because you're not actually in the cpp directory. Can you tell me what 'ls -al' shows where you are?

AhmedElsafy, 2018-04-19:
In the deploy_cpp directory

drwxrwxr-x 3 safy safy 4096 Apr 18 16:37 .
drwxrwxr-x 7 safy safy 4096 Apr 19 12:37 ..
-rw-rw-r-- 1 safy safy 2594 Apr 18 16:37 .clang-format
-rw-rw-r-- 1 safy safy 5018 Apr 18 16:37 README.md
drwxrwxr-x 5 safy safy 4096 Apr 18 16:37 src

safy@safy-MS-7B48:~/bonnet$ cd deploy_cpp/
safy@safy-MS-7B48:~/bonnet/deploy_cpp$ ls
README.md  src
safy@safy-MS-7B48:~/bonnet/deploy_cpp$ catkin init
Catkin workspace `/home/safy/bonnet` is already initialized. No action taken.
------------------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /home/safy/bonnet
------------------------------------------------------
Source Space:      [missing] /home/safy/bonnet/src
Log Space:         [missing] /home/safy/bonnet/logs
Build Space:       [missing] /home/safy/bonnet/build
Devel Space:       [missing] /home/safy/bonnet/devel
Install Space:      [unused] /home/safy/bonnet/install
DESTDIR:            [unused] None
------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------

------------------------------------------------------
WARNING: Your workspace is not extending any other
result space, but it is set to use a `linked` devel
space layout. This requires the `catkin` CMake
package in your source space in order to be built.

tano297, 2018-04-19:
It looks like you initialized another workspace inside bonnet as well. Can you run 'ls -al' in the /home/safy/bonnet directory?

tano297, 2018-04-23:
Were you able to solve this problem?

AhmedElsafy, 2018-04-24:
ROS re-installation solved the problem

closed 2018-04-24 by AhmedElsafy