PRBonn/bonnet #12: ./cnn_train.py -d cfg/cityscapes/data.yaml -n cfg/cityscapes/net_bonnet.yaml -t cfg/cityscapes/train_bonnet.yaml -l ../log/ -p ../pretrained

ytgcljj, 2018-04-24:
test16@test16:~/bonnet/train_py$ ./cnn_train.py -d cfg/cityscapes/data.yaml -n cfg/cityscapes/net_bonnet.yaml -t cfg/cityscapes/train_bonnet.yaml -l ../log/ -p ../pretrained
----------
INTERFACE:
data yaml:  cfg/cityscapes/data.yaml
net yaml:  cfg/cityscapes/net_bonnet.yaml
train yaml:  cfg/cityscapes/train_bonnet.yaml
log dir ../log/
model path ../pretrained
model type iou
----------

Commit hash (training version):  b'2ce83b0'
----------

Opening desired data file cfg/cityscapes/data.yaml
Opening desired net file cfg/cityscapes/net_bonnet.yaml
Opening desired train file cfg/cityscapes/train_bonnet.yaml
model folder exists! Using model from ../pretrained/iou
Copying files to ../log/ for further reference.
WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/tensorflow/contrib/learn/python/learn/datasets/base.py:198: retry (from tensorflow.contrib.learn.python.learn.datasets.base) is deprecated and will be removed in a future version.
Instructions for updating:
Use the retry module or similar alternatives.
Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Unable to init server: Could not connect: Connection refused
Failed to connect to Mir: Failed to connect to server socket: 没有那个文件或目录
Unable to init server: 无法连接： Connection refused

(cnn_train.py:7017): Gdk-CRITICAL **: gdk_cursor_new_for_display: assertion 'GDK_IS_DISPLAY (display)' failed
Traceback (most recent call last):
  File "./cnn_train.py", line 174, in <module>
    NET["name"] + '.py')
  File "/usr/lib/python3.5/imp.py", line 172, in load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 693, in _load
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 665, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/test16/bonnet/train_py/arch/bonnet.py", line 28, in <module>
    from arch.abstract_net import AbstractNetwork
  File "/home/test16/bonnet/train_py/arch/abstract_net.py", line 42, in <module>
    import dataset.aux_scripts.util as util
  File "/home/test16/bonnet/train_py/dataset/aux_scripts/util.py", line 22, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python3.5/dist-packages/matplotlib/pyplot.py", line 114, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/usr/local/lib/python3.5/dist-packages/matplotlib/backends/__init__.py", line 32, in pylab_setup
    globals(),locals(),[backend_name],0)
  File "/usr/local/lib/python3.5/dist-packages/matplotlib/backends/backend_gtk3agg.py", line 11, in <module>
    from . import backend_gtk3
  File "/usr/local/lib/python3.5/dist-packages/matplotlib/backends/backend_gtk3.py", line 58, in <module>
    cursors.MOVE          : Gdk.Cursor.new(Gdk.CursorType.FLEUR),
TypeError: constructor returned NULL

tano297, 2018-04-24:
This is a gtk problem, not a problem with the framework. If you are running this on ssh, you need to run it with the "-X" option. If you are running it locally, try updating matplotlib. Let me know if any of the options work

ytgcljj, 2018-05-02:
test@test:~/bonnet/train_py$ ./cnn_train.py -d data.yaml -n net.yaml -t train.yaml -l /tmp/path/to/log/ -p /tmp/path/to/pretrained
/home/test/.local/lib/python3.5/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
----------
INTERFACE:
data yaml:  data.yaml
net yaml:  net.yaml
train yaml:  train.yaml
log dir /tmp/path/to/log/
model path /tmp/path/to/pretrained
model type iou
----------

Commit hash (training version):  b'2ce83b0'
----------

Opening desired data file data.yaml
Error opening data yaml file. Check! Exiting...
test@test:~/bonnet/train_py$ ./cnn_train.py -d cfg/cityscapes/data.yaml -n   cfg/cityscapes/net_bonnet.yaml -t  cfg/cityscapes/train_bonnet.yaml -l /tmp/path/to/log/ -p /tmp/path/to/pretrained
/home/test/.local/lib/python3.5/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
----------
INTERFACE:
data yaml:  cfg/cityscapes/data.yaml
net yaml:  cfg/cityscapes/net_bonnet.yaml
train yaml:  cfg/cityscapes/train_bonnet.yaml
log dir /tmp/path/to/log/
model path /tmp/path/to/pretrained
model type iou
----------

Commit hash (training version):  b'2ce83b0'
----------

Opening desired data file cfg/cityscapes/data.yaml
Opening desired net file cfg/cityscapes/net_bonnet.yaml
Opening desired train file cfg/cityscapes/train_bonnet.yaml
model folder doesnt exist! Exiting...
test@test:~/bonnet/train_py$ ./cnn_train.py -d cfg/cityscapes/data.yaml -n   cfg/cityscapes/net_bonnet.yaml -t  cfg/cityscapes/train_bonnet.yaml -l /tmp/path/to/log/ -p /tmp/path/to/pretrained
/home/test/.local/lib/python3.5/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
----------
INTERFACE:
data yaml:  cfg/cityscapes/data.yaml
net yaml:  cfg/cityscapes/net_bonnet.yaml
train yaml:  cfg/cityscapes/train_bonnet.yaml
log dir /tmp/path/to/log/
model path /tmp/path/to/pretrained
model type iou
----------

Commit hash (training version):  b'2ce83b0'
----------

Opening desired data file cfg/cityscapes/data.yaml
Opening desired net file cfg/cityscapes/net_bonnet.yaml
Opening desired train file cfg/cityscapes/train_bonnet.yaml
model folder exists! Using model from /tmp/path/to/pretrained/iou
Copying files to /tmp/path/to/log/ for further reference.
WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/tensorflow/contrib/learn/python/learn/datasets/base.py:198: retry (from tensorflow.contrib.learn.python.learn.datasets.base) is deprecated and will be removed in a future version.
Instructions for updating:
Use the retry module or similar alternatives.
/usr/local/lib/python3.5/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
/usr/local/lib/python3.5/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
/usr/local/lib/python3.5/dist-packages/matplotlib/backends/backend_gtk3agg.py:18: UserWarning: The Gtk3Agg backend is known to not work on Python 3.x with pycairo. Try installing cairocffi.
  "The Gtk3Agg backend is known to not work on Python 3.x with pycairo. "
Training from model in  /tmp/path/to/pretrained/iou
Fetching dataset
Training with 4 GPU's
Training with batch size 20
DEVICE AVAIL:  /device:CPU:0
Number of GPU's available is 0
Traceback (most recent call last):
  File "./cnn_train.py", line 189, in <module>
    net.train(path=str(FLAGS.path + '/' + FLAGS.model))
  File "/home/test/bonnet/train_py/arch/abstract_net.py", line 1019, in train
    assert(self.n_gpus == self.n_gpus_avail)
AssertionError
test@test:~/bonnet/train_py$ ./cnn_use.py -l /tmp/path/to/log/ -p /tmp/path/to/pretrained -i /path/to/image
/usr/local/lib/python3.5/dist-packages/matplotlib/backends/backend_gtk3agg.py:18: UserWarning: The Gtk3Agg backend is known to not work on Python 3.x with pycairo. Try installing cairocffi.
  "The Gtk3Agg backend is known to not work on Python 3.x with pycairo. "
[libprotobuf FATAL google/protobuf/stubs/common.cc:61] This program requires version 3.5.0 of the Protocol Buffer runtime library, but the installed version is 2.6.1.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "external/protobuf_archive/src/google/protobuf/any.pb.cc".)
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  This program requires version 3.5.0 of the Protocol Buffer runtime library, but the installed version is 2.6.1.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "external/protobuf_archive/src/google/protobuf/any.pb.cc".)
\u5df2\u653e\u5f03 (\u6838\u5fc3\u5df2\u8f6c\u50a8)

ytgcljj, 2018-05-02:
test@test:~/bonnet/train_py$ ./cnn_use.py -l /tmp/path/to/log/ -p /tmp/path/to/pretrained -i /path/to/image
/usr/local/lib/python3.5/dist-packages/matplotlib/backends/backend_gtk3agg.py:18: UserWarning: The Gtk3Agg backend is known to not work on Python 3.x with pycairo. Try installing cairocffi.
  "The Gtk3Agg backend is known to not work on Python 3.x with pycairo. "
[libprotobuf FATAL google/protobuf/stubs/common.cc:61] This program requires version 3.5.0 of the Protocol Buffer runtime library, but the installed version is 2.6.1.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "external/protobuf_archive/src/google/protobuf/any.pb.cc".)
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  This program requires version 3.5.0 of the Protocol Buffer runtime library, but the installed version is 2.6.1.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "external/protobuf_archive/src/google/protobuf/any.pb.cc".)
已放弃（核心已转储）

tano297, 2018-05-02:
it seems like you are running into multiple problems at once:

- to fix the protobuf, and gtk problems, try updating these
```sh
$ sudo pip3 install -U tensorflow-gpu==1.7.0 protobuf==3.5.1 matplotlib==2.2.2
```
in order to train, you need a GPU, so considering that you get ```DEVICE AVAIL: /device:CPU:0```, I don't think there is much to do there. Do you have a gpu? If all cuda and cudnn drivers are properly installed, then the command above installing tensorflow for gpu should help.

The docker container has everything working, if you are having dependencies problems I suggest you give it a look

closed 2018-05-12 by tano297