PRBonn/bonnet #14: Retraining with Personal Data

AhmedElsafy, 2018-04-24:
I am trying to retrain with my own dataset, in the dataset/aux_script , it is mentioning to use "Use the output format extracted from the BAG that uses images and color labels created by Philipp's label creator."

Is this a kind of tool I should use first for my annotation?

tano297, 2018-04-25:
Hi,

Unfortunately, due to a cooperation with a company, the labeling tool that we use is proprietary, but here are some you can give a shot:

https://bitbucket.org/ueacomputervision/image-labelling-tool
https://github.com/tzutalin/labelImg
https://github.com/davidjesusacu/polyrnn-pp
http://is-innovation.eu/ratsnake/
https://rectlabel.com

I will close this as it is not related to the framework, and to avoid spamming everybody, but feel free to email me for non-framework related questions.

closed 2018-04-25 by tano297

AhmedElsafy, 2018-04-26:
Thanks Andres,

I am trying to retrain the network on my data, do you have any guide for
the data/labels format and structure to be stored in ?

like the RGB folders and train and test, the format of the labeled
segmented images ?

Thanks

tano297, 2018-04-27:
Hi,

I put a toy example in our server of the resulting dataset you get from pre-processing cityscapes (this case only one image) with the cityscapes parser included in the dataset folder, aux scripts.

http://ipb.uni-bonn.de/html/projects/bonnet/datasets/cityscapes_toy.tar.gz

By looking at this script, along with this dataset, and the data.yaml corresponding to cityscapes you should be able to understand better how the data format works!

AhmedElsafy, 2018-04-27:
Hi Andres,

I am just wondering for training the CWC model, do I have to convert my
segmented labeled images into mono-chrome images through the dataset,aux
scripts? or just load my data with colored masks and put the color map on
the data.yaml file

Thanks for your kind support

Regards