PRBonn/bonnet #15: Steps to use ROS node with docker image?

mfe7, 2018-05-09:
Hi,

I am using the docker image. I don't quite understand how to get the ROS node running.

```
$ roslaunch bonnet_run bonnet_run.launch
...
[ INFO] [1525824328.942834758]: Successfully launched node.
Unable to create network. 
Invalid yaml file /home/developer/Desktop/frozen_trial//train.yaml
[ERROR] [1525824328.942988464]: SOMETHING WENT WRONG INITIALIZING CNN. EXITING
[bonnet_node-2] process has died [pid 28742, exit code 1, cmd /home/developer/catkin_ws/devel/lib/bonnet_run/bonnet_node __name:=bonnet_node __log:=/home/developer/.ros/log/ae1c784a-531c-11e8-9312-38d547df6dab/bonnet_node-2.log].
log file: /home/developer/.ros/log/ae1c784a-531c-11e8-9312-38d547df6dab/bonnet_node-2*.log
```

It makes sense there is no model at that path, but I'm not sure how to proceed. In particular, I passed a model in through one of the args when starting docker as mentioned in the readme, but I don't know how to access this directory within the docker. If I just want to test a pre-trained model, does it make sense to download within the docker and freeze it?

I greatly appreciate all the documentation you have so far!

tano297, 2018-05-09:
Hi there,

- 1st: The way to change the pointer to frozen model and the topic names within the ROS node is by changing the 2 config files in `bonnet/deploy_cpp/src/ros/config/`. 

- 2nd: To understand where all the files are we need to break down the docker run command. In particular the following part:  `-v /home/$USER/data:/shared`. In the readme I wrote this is just an example. What -v means is that everything that is in your `/home/$USER/data` folder in the host will be accessible inside the running container in the `/shared` folder. 

What this means is that you need to modify this part of the command to point to wherever your models/datasets are in your host computer, and the config file to point to that frozen model.

As an example, let's say you have the pre-trained/frozen models in the directory `/home/$USER/Downloads`. Then if you modify the full docker run command to be:

```sh
$ nvidia-docker run -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v $HOME/.Xauthority:/home/developer/.Xauthority -v /home/$USER/Downloads:/hostDownloads --net=host --pid=host --ipc=host bonnet /bin/bash
```
Now your data will be inside the container in `/hostDownloads`, which means that in your config file you will need to make:

```yaml
model_path: "/hostDownloads/NAMEOFFROZENFOLDER"
device: "/gpu:0"
verbose: true
backend: "tf" # tf (tensorflow, cpu or gpu mode) or trt (TensorRT nvidia gpu)
```

If you have the pretrained models in Downloads, then before you use the ros node you will need to run (inside docker):

```ssh
$ cd train_py
$ cnn_freeze.py -p /hostDownloads/PRETRAINEDDIR -l /hostDownloads/NAMEOFFROZENFOLDER
```

I hope it is clear now! Let me know if I am missing something.

mfe7, 2018-05-09:
This is great! Thanks for the thorough description - it is much more clear
now. It may be a good idea to add this info to one of the readmes

closed 2018-05-12 by tano297

daobilige-su, 2018-09-25:
Hi @mfe7 @tano297 

I am a bit new to docker, so after I pulled the docker image (tano297/bonnet:cuda9-cudnn7-tf17-trt304), I don't know how to build it using (nvidia-docker build -t bonnet .) as stated in the readme file.

when I go in to the docker image without building it, I clearly do not has access to functions like ros etc.

Could any of you suggest me how to deal with it? Thanks.

Su

tano297, 2018-09-29:
Hello @daobilige-su ,

what do you mean by `when I go in to the docker image without building it`? You need to build the container to run it. 

The steps are build (you need to be IN the root directory of Bonnet, where the Dockerfile is):
```
$ nvidia-docker build -t bonnet .
```

And then run with an interative command line:
```
$ nvidia-docker run -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v $HOME/.Xauthority:/home/developer/.Xauthority -v /home/$USER/data:/shared --net=host --pid=host --ipc=host bonnet /bin/bash
```
The last step will put you in a cmd inside the container image and will allow you to do anything you can do in an ubuntu with all the dependencies installed and the Bonnet files there. So you can create a catkin workspace and build bonnet without worrying about installing Tensorflow or tensorRT

tano297, 2018-09-29:
Hi there,

The reason why you don't need to build it is that in your command you are using the `tano297/bonnet:cuda9-cudnn7-tf17-trt304` image from my docker hub.
The dockerfile contained in this repo, which builds into the `bonnet` image has some extra dependencies installed, and the modification of some environment variables that make tensorflow and tensorRT run as well. I'm surprised you were able to run the scripts.

I'm glad it works, but if you run into any problems in the future, I recommend using the dockerfile included here, so that the extra steps needed are taken care of, by using the 2 steps above.

Best, Andres