PRBonn/bonnet #16: how do you use the 14 channels images to the CNN?

wenjuntan, 2018-05-18:
Hello，I have read your paper, in your paper, the 14 channels image are used, but, what should I do to achieve it？

tano297, 2018-05-22:
The implementation of the extra representations is in the codebase, but in order to make the framework more general, I had to strip it out of the multi-dimensional support. I was working on a way to put in the possibility in a nice way again, but I haven't had time. It should not be hard to add again, so if you want to create a pull request with an initial estimate, do so, and we can figure it out together.

closed 2018-05-22 by tano297

louisflorentin, 2020-10-14:
Hi, thanks for your GREAT framework, it is really useful for me. I am trying to implement the 14 channels and I would like to know what parameters you chose in the model (14 channels) you present in your article. Could you please share your cfg files (net.yaml, train.yaml) ? I attached my cfg files, which don't lead to an efficient model. Thank you very much.

[image]
[image]