PRBonn/bonnet #17: roslaunch bonnet_run bonnet_run.launch error?

ytgcljj, 2018-05-18:
This is the docker I run,
i run  roslaunch bonnet_run bonnet_run.launch 

Unable to create network. 
Invalid yaml file /shared/city_512//nodes.yaml
[ERROR] [1526639307.186656472]: SOMETHING WENT WRONG INITIALIZING CNN. EXITING
[bonnet_node-2] process has died [pid 27831, exit code 1, cmd /home/developer/catkin_ws/devel/lib/bonnet_run/bonnet_node __name:=bonnet_node __log:=/home/developer/.ros/log/32c8b5d4-5a86-11e8-b58d-507b9deaef1f/bonnet_node-2.log].
log file: /home/developer/.ros/log/32c8b5d4-5a86-11e8-b58d-507b9deaef1f/bonnet_node-2*.log

tano297, 2018-05-22:
it looks like you are having some trouble with the frozen model. 
Did you freeze the downloaded model? Check out how to do it [here](https://github.com/PRBonn/bonnet/tree/master/train_py).

You would need to run something like:
```sh
$ ./cnn_freeze.py -p /shared/city_512/ -l /shared/city_512_frozen/
```

ytgcljj, 2018-05-23:
i run ./cnn_freeze.py -p /shared/city_512/ -l /shared/city_512_frozen/

Predicting mask 
mask shape [1, 256, 512]
Restoring checkpoint
Looking for model in /shared/city_512/iou
Retrieving model from:  /shared/city_512/iou/model-best-iou.ckpt
Successfully restored model weights! :D
Saving this graph in /shared/city_512_frozen/
Converted 215 variables to const ops.
Saving uff model for TensorRT inference
Using output node test_model/model/logits/linear/BiasAdd
Converting to UFF graph
No. nodes: 525
UFF Output written to /shared/city_512_frozen/optimized_tRT.uff
Killing all threads and exiting!
Exception ignored in: <bound method BaseSession.__del__ of <tensorflow.python.client.session.Session object at 0x7f83d5eb4320>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/client/session.py", line 712, in __del__
TypeError: 'NoneType' object is not callable

tano297, 2018-05-26:
Did you change the path in the launch file after that? If you are getting another error, please post it, otherwise I don't have enough data to help

closed 2018-06-03 by tano297

daobilige-su, 2018-09-27:
Hi @ytgcljj ,

how did you run ` roslaunch bonnet_run bonnet_run.launch` inside docker image? I could not find the bonnet package inside the docker image, so I installed in manually. However I am still getting error with no clue after launching bennet_run.launch.

Any help is appreciated.

Su