PRBonn/bonnet #18: Invalid yaml file persons_512/nodes.yaml

chenyumingx, 2018-05-22:
I have encountered the same problem:
log:
Successfully created log directory: log
Unable to create network.
**Invalid yaml file persons_512/nodes.yaml**

I have tried "./cnn_freeze.py -p /home/chen/log" to freeze the log folder,however another problem occured: 
**Traceback (most recent call last):
  File "./cnn_freeze.py", line 32, in <module>
    import tensorflow as tf
ImportError: No module named 'tensorflow'**

but I do not fully understand your answer, and **do not know how to solve it**. Can you answer in more details? Thank you very much!

tano297, 2018-05-26:
In the second case, the problem is self explanatory, you need to install tensorflow

```
$ sudo pip3 install tensorflow tensorflow-gpu
```

Have you tried doing it using the docker container? Its main purpose is to give you all of the dependencies solved. You can freeze it with docker and then use it in your host OS, which apparently has everything for the C++ library because it builds

chenyumingx, 2018-05-26:
I successfully installed tensorflow
I still don't know how to freeze the model,I will try to use the docker container

tano297, 2018-05-26:
Hi, what do you mean by how? Which command? Or are you still getting errors? If your are getting errors, you can post them here, but with the docker container, the instructions and the pretrained model, you should be fine

closed 2018-06-03 by tano297