PRBonn/bonnet #20: Jetson TX1

181charan, 2018-06-03:
I was able to run the inference using the python files on a host PC and on the Jetson. However I could not run the Tensorrt optimised cnn_use_pb_tRT on the Jetson because Tensorrt python libraries aren't available for the arm64 environment of Jetson. Just to clarify to actually compile the deploy cop
1) build Basel from source for arm64
2) Build the tensorflow_cc external install for arm64
3) Then use catkin to build the executable right?
Thank you.

tano297, 2018-06-03:
Hi there, if you want to use with the tensorRT backend in C++ you don't need to compile the tensorflow backend.

Can you put this repo in a catkin workspace and compile it to see what CMake outputs? It is strange that the python tensorRT files are not working for you, because jetpack comes with all of these things pre-installed.

181charan, 2018-06-03:
This is my output when im trying to compile on the Jetson. I tried to compile this on my other PC and got the EXACT same error though.

`nvidia@tegra-ubuntu:/media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp$ catkin build bonnet_standalone
----------------------------------------------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp
----------------------------------------------------------------------------------
Source Space:       [exists] /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/src
Log Space:         [missing] /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/logs
Build Space:        [exists] /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/build
Devel Space:        [exists] /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/devel
Install Space:      [unused] /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/install
DESTDIR:            [unused] None
----------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------------------------------
NOTE: Forcing CMake to run for each package.
----------------------------------------------------------------------------------

----------------------------------------------------------------------------------
WARNING: Your workspace is not extending any other result space, but it is set to
use a `linked` devel space layout. This requires the `catkin` CMake package in
your source space in order to be built.
----------------------------------------------------------------------------------

[build] Found '3' packages in 0.0 seconds.                                                          
[build] Updating package table.                                                                     
Starting  >>> catkin_tools_prebuild                                                                 
____________________________________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/logs/catkin_tools_prebuild/build.cmake.000.log
CMake Error at /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/build/catkin_tools_prebuild/CMakeLists.txt:12 (message):
  The catkin CMake module was not found, but it is required to build a linked
  workspace.  To resolve this, please do one of the following, and try
  building again.

   1. Source the setup.sh file from an existing catkin workspace:
      source SETUP_FILE

   2. Extend another catkin workspace's result (install or devel) space:
      catkin config --extend RESULT_SPACE

   3. Set `catkin_DIR` to the directory containing `catkin-config.cmake`:
      catkin config --cmake-args -Dcatkin_DIR=CATKIN_CMAKE_CONFIG_PATH

   4. Add the catkin source package to your workspace's source space:
      cd SOURCE_SPACE && git clone https://github.com/ros/catkin.git

cd /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/media/nvidia/BF0E-10D0/new/bonnet/deploy_cpp/install; cd -
....................................................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]                           
Failed    <<< catkin_tools_prebuild                [ 10.9 seconds ]                                 
Abandoned <<< bonnet_core                          [ Unrelated job failed ]                         
Abandoned <<< bonnet_standalone                    [ Unrelated job failed ]                         
[build] Summary: 0 of 3 packages succeeded.                                                         
[build]   Ignored:   1 packages were skipped or are blacklisted.                                    
[build]   Warnings:  None.                                                                          
[build]   Abandoned: 2 packages were abandoned.                                                     
[build]   Failed:    1 packages failed.                                                             
[build] Runtime: 11.2 seconds total.                                                                
Exception ignored in: <bound method BaseEventLoop.__del__ of <_UnixSelectorEventLoop running=False closed=True debug=False>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/trollius/base_events.py", line 395, in __del__
  File "/usr/local/lib/python3.5/dist-packages/trollius/unix_events.py", line 65, in close
  File "/usr/local/lib/python3.5/dist-packages/trollius/unix_events.py", line 166, in remove_signal_handler
[... cut ...]
sidenote: Do you have any tips on setting up the cityscapes dataset for  training? Things I did were, download the cityscapes dataset and use the cityscapes_preprocess.py from the aux scripts. Then I got an error during training which said "Input tensor (1,) but tensor must be (2,?,?,3)"  in function 'print('Training Data Eval:')
          ignore_last = self.TRAIN["ignore_crap"]
          m_acc, m_iou, _, _, _ = self.training_dataset_accuracy(
              self.dataset.train, self.batch_size, self.batch_size_gpu, ignore_last)' in the file abstract_net.py

tano297, 2018-06-04:
Hi,
In terms of the C++ cmake problem, can you try `$ sudo apt-get install python-catkin-pkg`
For cityscapes training, it is hard to say why you would get that error. If the images and labels were created by the cityscapes_preprocess script they should work. Are you using a custom architecture or are you using one of the provided ones?

tano297, 2018-06-18:
Have you solved this problem? I will close this thread, if it is still not working, reopen

closed 2018-06-18 by tano297