PRBonn/bonnet #21: ros not catkin_make

ytgcljj, 2018-06-06:
CMakeFiles/Makefile2:445: recipe for target 'bonnet/src/lib/CMakeFiles/bonnet_core.dir/all' failed
make[1]: *** [bonnet/src/lib/CMakeFiles/bonnet_core.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 19%] Generating C++ code from darknet_ros_msgs/CheckForObjectsGoal.msg
[ 19%] Generating Lisp code from darknet_ros_msgs/CheckForObjectsActionGoal.msg
[ 20%] Generating Lisp code from darknet_ros_msgs/CheckForObjectsFeedback.msg
[ 20%] Built target usb_cam
[ 21%] Generating Lisp code from darknet_ros_msgs/CheckForObjectsActionResult.msg
[ 21%] Generating Lisp code from darknet_ros_msgs/CheckForObjectsGoal.msg
[ 21%] Generating C++ code from darknet_ros_msgs/CheckForObjectsAction.msg
[ 23%] Generating Lisp code from darknet_ros_msgs/CheckForObjectsAction.msg
[ 23%] Built target darknet_ros_msgs_generate_messages_lisp
[ 23%] Built target darknet_ros_msgs_generate_messages_cpp
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

tano297, 2018-06-06:
Hi there, unfortunately I can't make much out of this, most of the message is for other packages.
Is there a reason for using catkin_make? If you are mixing it with catkin build this can cause problems. I would suggest you start from a clean build. Beware that this will also clean your other packages, so it may take a while to build if you have many of them.

```sh
$ catkin clean -y
$ catkin build
```

If it still doesn't work, please provide more information. The build is currently working in the CI so it has to be a problem with your setup

tano297, 2018-06-18:
Closing this, if you could not fix it, reopen

closed 2018-06-18 by tano297