PRBonn/bonnet #22: own data preparation

wzhouuu, 2018-06-07:
Hi, thank you for the great work! I've successfully tried your trained model in ROS, it looks great!

Just a few questions to start training my own model:
1. Is the whole pipeline working with lower cuda like cuda8 and cudnn5? Also my tensorflow version is 1.4

2. Is that possible to provide more instructions on how to train on own dataset? I have my dataset ready, which has images and labels (class number from 0-11). So I don't need any remap work but just need to rearrange them to match the format you provided in Issue #14 . However if I'd like to use your augmentation pipeline, how should I do it? Like which file I should run, what command I need, etc. 

3. Based on my understanding, I presume every time train from scratch, the train/test/valid sets will be shuffled and changed, that's why need log folder to save the training parameters?

Thanks again for the help!

wzhouuu, 2018-06-07:
More in detail, I wrote the data.yaml as below (change to txt otherwise couldn't upload)
[data.txt](https://github.com/PRBonn/bonnet/files/2078850/data.txt)
In my data dir, I have train, test, valid folder. In each of them, I have img and lbl folders to save images and labels. 

Then I run `./cnn_train.py -d cfg/cs_my/data.yaml -n cfg/cs_my/net_bonnet.yaml -t cfg/cs_my/train_bonnet.yaml -l /media/data/bonnet_result/cs_result/log`

Then it said `parsing directory`, `Image file xxx.png` has no label, GIMME DAT LABEL YO! Ignoring...
What's the correct step to load data?

tano297, 2018-06-08:
Hi there,

`Is that possible to provide more instructions on how to train on own dataset? I have my dataset ready, which has images and labels (class number from 0-11). So I don't need any remap work but just need to rearrange them to match the format you provided in Issue #14 . However if I'd like to use your augmentation pipeline, how should I do it? Like which file I should run, what command I need, etc.`

The augmentation is included in the way we load the data. Currently, horizontal flips, gamma ablations, and gaussian blurs are applied, because they are the only ones that make sense for all datasets. If you need more, you can easily add it [here](https://github.com/PRBonn/bonnet/blob/7c7a1641580d989f275aba5a447c8b7d85fd08cf/train_py/dataset/abstract_dataset.py#L43)

`Do you have a general idea how long the training may take if I have one 1080Ti GPU and 10k images?`

Synthia is approximately like that, and with the proper hyperparameters it gets to accuracy overnight, if starting from a pretrained model. But it depends more on the data than any other thing. If your images are highly correlated, convergence is faster. Otherwise you need more epochs

`I guess all the data loaders assume img and lbl have same name? otherwise need preprocess to save them as same name?`

yes, the general parser expects pngs with same name for corresponding images and labels, and then it generates remapped directories where the images are converted to jpeg and resized to the input size of the cnn for faster opening during training

wzhouuu, 2018-06-14:
I finally trained my own model and it looks pretty good! 
I used ENet before, I think the way to load CamVid dataset is quite neat. just need text files to list images and labels. So the images do not need to be rearranged into different folders when we want to re-split train/val/test. Anyway, thanks for the help!

closed 2018-06-14 by wzhouuu