PRBonn/bonnet #30: validation matrix

cshort101, 2018-08-17:
Hi,
I'm wondering if there's any confusion matrix output from either training or validation to see the per class accuracy, IoU, etc?

tano297, 2018-08-26:
Hi, sorry for the delay, I've been out and about.

Using the [cnn_dataset_performance.py](https://github.com/PRBonn/bonnet/blob/master/train_py/cnn_dataset_performance.py) script you can get the classwise metrics of an already trained model. The confusion matrix is not published, but it is calculated, so you should be able to print if by making the output of the IoU calculation a bit more verbose, if you need this I can tell you where to look.

closed 2018-08-26 by tano297

cshort101, 2018-08-27:
Thanks for the reply! IoU is quite important for me to compare different networks, so could you please provide some hints for that part?