PRBonn/bonnet #32: Multi-input multi-output model

zaher88abd, 2018-09-05:
This is not an issue, it is a question.
There is a fast way to create a Multi-input multi-output model because I read the code and I found it needs to create a new abstract_net class. please, correct me if I am wrong.

tano297, 2018-09-06:
It depends on what you are trying to do. If the inputs and outputs are all images of the same size you may be able to squeeze them both into the same tensor and just modify the architecture to unpack the inputs after receiving and pack them before sending out. 

I think otherwise it would take a considerable amount of effort to implement, especially in the C++ inference part, since this case was not remotely considered. May I ask what the particular application is (inputs/outputs)? Maybe I can give you a hand.

zaher88abd, 2018-09-14:
Hi @tano297,
Thanks for the reply. I am thinking to use your framework testing model has two images as input and two images as output. I will try your idea. 
Thanks.

closed 2018-09-14 by zaher88abd