PRBonn/bonnet #33: ros catkin build can't find TensorRT

cshort101, 2018-09-06:
Hi, I was trying to deploy bonnet on NVIDIA DRIVE PX2. I think TensorRT is already installed for PX2. But when I tried to catkin build, it complained didn't find Tensorflow or TensorRT.. Is there any specific setup I need to do to fix this problem?

[image]

tano297, 2018-09-06:
Hi there. I haven't made the move to tensorRT4 yet, this may be the problem. Can you try putting some traces in the cmake file? 

I don't only look for libnvinfer, but also libnvparsers:

```
find_library(NVINFER NAMES nvinfer)
find_library(NVPARSERS NAMES nvparsers)
```

I don't have access to an embedded nvidia computer now (drive or jetson), but I am getting a jetson xavier as soon as they start dispatching, so until then you will need to help me debug it remotely to make it work if you need it. Otherwise, I would suggest installing an old jetpack with tensorrt3 which I know works, but it's a considerable amount of effort if you have lots of other programs installed in the device

cshort101, 2018-09-07:
Sorry it's my bad, the tensorrt wasn't installed properly in our PX2. the current default version is still 3.0.2-1. I've successfully compiled Bonnet in ROS in our PX2.. sorry for the trouble!

closed 2018-09-07 by cshort101

thanhngotud, 2018-11-25:
Hi @cshort101, I also trying to build bonnet in PX2 also. I very appreciate if you could share the step to install Bonnet in the PX2? Thanks

cshort101, 2018-11-25:
@thanhngotud I don't think I did anything special. If you have ROS (self compiled) and TensorRT (automatically installed) in Nvidia PX2, you should be able to catkin_make bonnet directly...

thanhngotud, 2018-11-25:
@cshort101 I'm sorry to ask something not relevant with bonnet, but did you install ROS by cross-compile or directly in the PX2?

cshort101, 2018-11-25:
directly in PX2