PRBonn/bonnet #36: running docker image

daobilige-su, 2018-09-26:
Hi,

First, thanks for the software. Looks very cool.

I am using the docker image provided. But I have 2 questions about it.

After I read the dockerfile in the image, if I understood correctly, all dependencies are built in the image, but the actual **bonnet** is not installed inside the image. Is that correct? Because I could not find the lines correspond to installation of bonnet code. If so, do I have to install it by myself on top of the image?

I ran the **helloworld.py** under the \bonnet-docker folder of the image. Then I got **Segmentation fault (core dumped)** error. When I execute the code inside **helloworld.py** line by line, I come to know that it is the **import tensorrt** causing the error. Does it works fine on you machine?

Thanks for any help or suggestion.

Cheers,
Su

tano297, 2018-09-29:
Hello,

I am not having any problems running the hello world. Can you check that you are using the proper docker version? You should be using nvidia-docker [link](https://github.com/NVIDIA/nvidia-docker), not the vanilla one. 

You can check if your gpu functions are working inside the docker container running

```sh
$ nvidia-smi
```

daobilige-su, 2018-09-29:
Hi @tano297 ,

Thanks for your reply.

I finally made everything running now. It turns out that I need to delete /usr/local/cuda/lib64/stubs/libcuda.so.1 file to make tensorrt and tensorflow work. Also I need to recompile tensorflow C++ API by adding CC_OPT_FLAGS="-march=native" flag before compiling to support my CPU version. 

It is a really nice software, enjoying it now. Thanks.

Cheers,
Su

daobilige-su, 2018-10-18:
Hi,

First, you need to make sure the problem arise from tensorflow C++ API. To do that, just run the test program of it.

```
$ cd /tools/tensorflow_cc/example
$ mkdir build && cd build
$ cmake ..
$ ./example
```

If the above test program is giving you the same error, then it is surely the tensorflow C++ API is the source of the error. Your CPU version is too old to be supported by the default configuration of tensorflow C++ API. To recompile tensorflow API, do followings:

```
$ cd /tools/tensorflow_cc/tensorflow_cc
$ mkdir build
$ cd build
$ export CC_OPT_FLAGS="-march=native"
$ cmake -DTENSORFLOW_STATIC=OFF -DTENSORFLOW_SHARED=ON -DTENSORFLOW_TAG="v1.7.0" ..
$ make -j
$ make install
$ rm -rf ~/.cache && cd .. && rm -rf build
```

after that, you might also needs to re-install tensorflow again, since the installation of tensorflow C++ API will install a different version of tensorflow, which you have to uninstall and install the correct version of tensorflow again.

```
RUN pip3 uninstall numpy tensorflow-gpu tensorflow matplotlib -y && \
pip3 install -U tensorflow-gpu==1.7.0 protobuf==3.5.1 matplotlib==2.2.2
```

Hopefully that's it. 

Cheers,
Su

hyejun, 2018-10-24:
I checked problem arise from tensorflow C++ API.
Then, I tried to install tensorflow again.
But, there are some errors while building tensorflow.

So, I tried to install docker to another computer and it is ok.
you said " Your CPU version is too old to be supported by the default configuration of tensorflow C++ API. ", maybe it is right.

Thank you for answering.

blubbi321, 2018-11-03:
I can confirm the issue. No problems following along the instructions on a more recent machine. However, I could not yet resolve all the dependencies for the steps @daobilige-su mentioned above. (Apparently one needs to also install g++-7, which is then in turn incompatible with the cuda libs "/usr/local/cuda-9.0/bin/../targets/x86_64-linux/include/crt/host_config.h:119:2: error: #error -- unsupported GNU version! gcc versions later than 6 are not supported!
 #error -- unsupported GNU version! gcc versions later than 6 are not supported!")

Machine with the trouble is an Intel i7-2600K in case that helps anybody