PRBonn/bonnet #38: the output is not right

zaher88abd, 2018-10-19:
Hi,

I would ask if you can know the reason of why there are points on predication output 

[image]

the points like row and columens.

Thanks.

tano297, 2018-10-22:
Hi.
It is common to see those checkerboard artifacts for some decoder architectures, but not for the ones that are included. Can you tell me which setup you are using?

zaher88abd, 2018-10-22:
Thank you @tano297 for the response, the issue is model needed for more training. I got this image after 100 epoch, and it confused me because of the pattern of green points in the image. However, the pattern has gone after 500.
In any way, there is big changing in Tensorflow coming in V2.0; So, are you interested in changing this framework to Keras or Next Tensorflow?

tano297, 2018-10-22:
Hello again,

We will see how much the effort is to change to Tensorflow2. According to the changes from the manifesto, it should be trivial to adapt the tensorflow part to make it work. However, the main purpose of Bonnet is to allow for fast inference using C++ and TensorRT, so as long as TensorRT doesn't support tensorflow2 generated graphs, we will have to stick to Tensorflow1.x. 
According to the information we have so far, it shouldn't change, but tensorRT moves always a bit behind the main libraries so we will see when everything is launched.

I tried inference of the person segmentator on the new tensorRT with an NVIDIA Jetson Xavier last week, and everything so far is running smoothly.

zaher88abd, 2018-10-22:
Me too I don't have much information about Tenosorflow 2.x, but I saw this [video](https://www.youtube.com/watch?v=WTNH0tcscqo&t=46s)  by Aurélien Géron explain a little bit about it. 
I tried Bonnet on Jeston before, which is good because it is using a few power, but it is very slow; I just got 4 to 6 frames per second for prediction.
Can I know, how many frames you got?

tano297, 2018-10-22:
The FPS depends on the model and the input size, which depends on the task.
For Crop-Weed, the model size and input size are small enough to run it through the TensorRT backend very fast (more than 15fps). For cityscapes models, this changes because the models are bigger and so are the inputs if you want fine-grained detail. There is also a huge difference in performance between the jetson TX2 and the Xavier AGX. If you share with me the implementational details we can discuss it, but I would suggest that you optimize the input resolution to be the minimum that gives you good results, and using the tensorRT backend for inference. 

I am closing this now because the original issue was solved, but message me for any further questions

closed 2018-10-22 by tano297

zaher88abd, 2018-10-22:
Thanks, I used model Crop-weed with images size (360, 512) and yes Jetson I have is TX2. However, thank you again for nice chatting.
Please, in future if you are looking for help to improve this framework let me know, I would like to help.