PRBonn/bonnet #43: ros catkin build Error in netTRT.cpp

LukiMuc, 2019-01-08:
Hallo,

I tried to build the package in ROS Kinetic on Jetson TX2 (Jetpack 3.3) and received following error:
[image]

Do anyone know why I receive this message?

Thanks

tano297, 2019-01-09:
I'm not able to test it right now, but I had trouble with that API call in 3.3 and I created a branch "jetpack4.1-jetson-xavier" a couple of months ago. Can you test this branch to see if it is solved? Otherwise I will have to have a deeper look into it.

LukiMuc, 2019-01-09:
With the new branch for Jetpack4.1 it builded with catkin.

Thank you! :+1:

closed 2019-01-09 by LukiMuc