PRBonn/bonnet #45: error: no matching function for call to ‘nvuffparser::IUffParser

JadBatmobile, 2019-02-01:
Hello,

I am attempting to use deploy_cpp with TensorRT backend.

I have installted TensorRT 4 succesfully, 

when i run catkin_make, it throws the warning that i do not have tensorflow_cc, but that tensorRT is found. I am under the impression that should be enough?

The following is what i am seeing 
/home/ubuntu/bonnet_ws/src/bonnet/deploy_cpp/src/lib/src/netTRT.cpp:113:56: error: no matching function for call to ‘nvuffparser::IUffParser::registerInput(const char*, nvinfer1::DimsCHW&)’
   _parser->registerInput(_input_node.c_str(), inputDims);

JadBatmobile, 2019-02-01:
I tried TensorRT 5 first actually, and then i thought the function signature may have changed for 5, so i went to TensorRT4 and the same occurred

JadBatmobile, 2019-02-01:
is the the signature it is supposed to have: 
virtual bool registerInput(const char* inputName, nvinfer1::Dims inputDims, UffInputOrder inputOrder)?

should i true tensorRT 3?

tano297, 2019-02-01:
Hi,

Have you tried the branch `jetpack4.1-jetson-xavier`?
I haven't had time to merge the branches and compile different api calls depending on the version of tensorRT. Actually it's pretty depressing that NVIDIA would make such a change, it seems unnecessary. I will add an issue so I remember to fix this.

closed 2019-02-05 by tano297

ToniRV, 2019-04-11:
I see the same issue with `TensorRT5`, I switched to the suggested branch, and solves the issue. Thanks