PRBonn/bonnet #47: input_norm_and_resized_node vs input_node

JadBatmobile, 2019-02-05:
Hello,

I am curious what the difference between the input_node and the input_norm_and_resized_node are?

I see in deploy_cpp that netTF.cpp utilizes input_node, and netTRT.cpp utilizes input_norm_and_resized_node.

JadBatmobile, 2019-02-05:
i see that one is raw image, and the other is resized and normalized, but why is it necessary that TRT accepts the processed version, and TF the raw version?

tano297, 2019-02-05:
Unfortunately some of the operators between the input and the norm_resized version are not supported by tensorRT. At least at the time that I built the inference pipeline. If you want to try to change that and it works with newer tensorRT versions, I'm more than happy accepting a pull request :)

JadBatmobile, 2019-02-05:
I understand, which tensorRT version are you using? it is not specified in the README

JadBatmobile, 2019-02-05:
Also, in bonnet.py, lines 52 - 62, i see img_pl being resized with tf.image.resize_images, then transposed with tf.transpose, then normalized with a regular python subtraction and division ( - 128 / 128 ). Do you recall which of those operations specifically was not running in TensorRT?

tano297, 2019-02-05:
It was definitely the resizing. And in older versions of tensorRT also division was a problem given some nomenclature and API problems. The division seems to be supported in tensorRT5, but resizing I don't think. In my computer I'm currently running tensorRT3 and in my jetson AGX tensorRT5, and it's working for both in the current state

JadBatmobile, 2019-02-05:
Okay, thank you for the speedy responses

JadBatmobile, 2019-02-05:
If those preprocessing steps are not being done by TensorRT, are they being done on the processor rather than the GPU?

closed 2019-02-05 by JadBatmobile