PRBonn/bonnet #52: Installation on Jetson TX2

dmicha02, 2019-05-20:
Hi,

I would like to use Bonnet on a Jetson TX2 to do inference.
I try to install it with docker (by using indicated procedure) but it doesn't work.
Someone has already succeed in this task?

Thank you,

tano297, 2019-05-21:
Hello,

Docker on the Jetson doesn't make too much sense because most of the dependencies are already installed on the jetson. Have you tried just running the C++ part directly on the jetson as it comes?

dmicha02, 2019-05-22:
I'm beginner with Jetson, docker and ROS, so sorry for my stupid question.
I follow the procedure describe in Readme in deploy_cpp part and it works perfectly by using JetPack 3.2 (with tensorRT 3) and https://github.com/jetsonhacks/installROSTX2 for the installation of ROS Kinetic.

Thanks,

closed 2019-05-22 by dmicha02