PRBonn/bonnet #55: docker ros key expired

weizhou1001, 2019-08-22:
I was trying to build other stuff in your docker image. But the ROS complaining: 
Err:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-pcl-msgs amd64 0.2.0-0xenial-20180222-204939-0800  404  Not Found [IP: 140.211.166.134 80]

I think the ROS key in the docker is expired. So I did:
$ sudo apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116
$ sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
$ sudo apt clean && sudo apt update

It fixed the problem.. Just wanna double check if this is an issue when you use it?

tano297, 2019-08-22:
Yes, this is the case. 
I will try to update the base image to reflect this, and leave this issue open just in case it happens to anybody else

weizhou1001, 2019-08-23:
Also want to check, after I run ```  $ nvidia-docker run -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v $HOME/.Xauthority:/home/developer/.Xauthority -v /home/$USER/data:/shared --net=host --pid=host --ipc=host bonnet /bin/bash```, which command should I use to go back to that container or open multiple terminals with the same setting? I found if I still run the same command, it will clean my previous setup...

tano297, 2019-08-23:
Hello @weizhou1001 

I don't think there is a straight forward way to do that. If you are not worried about security, I would just make a more general shared folder where you have all your data, and use that at the beginning. 

This is a very specific corner case that is not related to Bonnet, but docker itself, so here are some resources:
- run vs exec / image vs container: https://spin.atomicobject.com/2018/10/04/docker-command-line/
- multiple container terminals: https://stackoverflow.com/questions/39794509/how-to-open-multiple-terminals-in-docker
- saving state of container as image for reuse: https://docs.docker.com/engine/reference/commandline/commit/

aravind93raja, 2019-09-06:
I am getting the same issue in Docker : 
When I run sudo apt-get update

W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros/ubuntu xenial InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Failed to fetch http://packages.ros.org/ros/ubuntu/dists/xenial/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Some index files failed to download. They have been ignored, or old ones used instead.

How do I fix this ?

tano297, 2019-09-06:
Hi @aravind93raja ,
Although it is a related problem, it is not the same, since the key is different.
As a general rule, please first try googling, since this is not an issue with Bonnet.
This may help you: [link](https://chrisjean.com/fix-apt-get-update-the-following-signatures-couldnt-be-verified-because-the-public-key-is-not-available/)