PRBonn/bonnet #64: How to get datasets

lccatala, 2020-07-20:
Hi, I'm trying to run this but can't figure out how to add the required datasets. Already tried it with the Cityscapes one, but it has a different structure than the one the system seems to require. Am I supposed to download them from somewhere and put them in a particular folder in a specific way?

tano297, 2020-07-20:
https://github.com/PRBonn/bonnet/blob/master/train_py/dataset/aux_scripts/cityscapes_preprocess.py 

you have to use that script

tano297, 2020-07-20:
I suggest using this repo, as it is more up to date though...
https://github.com/PRBonn/bonnetal

lccatala, 2020-07-21:
Thanks! (I'm using this repo since it's the one my professors recommended, might try the new one afterwards). I've also found it seems the script https://github.com/PRBonn/bonnet/blob/master/train_py/dataset/aux_scripts/cityscapes_preprocess.py is broken (line 213), since it picks up every image ending in "*_gtFine_labelTrainIds.png" as a label, and there are not such files in the dataset, thus giving me an empty /lbl/ folder. the only tails present are "*_gtFine_labelIds.png" and "*_gtFine_instanceIds.png". Should I change it to one of those? And in that case, to which one?

tano297, 2020-07-21:
Before you can use the script in bonnet you need to create these labelids.png images using the scripts from the [cityscapes API](https://github.com/mcordts/cityscapesScripts/tree/master/cityscapesscripts/preparation). You need to map the classes yourself using the dictionary in [this file](https://github.com/mcordts/cityscapesScripts/blob/84b6ecef6d281acb6f67119e50fbd153c54977af/cityscapesscripts/helpers/labels.py#L62) to make them match my yaml file

closed 2020-07-21 by tano297

lccatala, 2020-07-23:
Should the images only contain two classes? After preprocessing them they only have pixels for road and sidewalk, with most of them untagged (and it drops all of them)